riclolsen/json-scada #3: Open port for demo json:scada

joyshmitz, 2020-09-29:
Hi, Ricardo what ports should be open for demo json:scada

riclolsen, 2020-09-29:
Hi,
The online demo was stopped, probably this was the problem as the docker demo tries to connect to it.
Now it is running, please retry.
You only need the Internet connection and the ability to access port 2404 (outwards) on the online demo server.
You can connect with an IEC 104 client to the local demo IEC104 server on localhost 2404.

joyshmitz, 2020-09-30:
thanks for the quick response.

I want to help a great project,
I can work on a solution for monitoring Docker hosts and containers using Prometheus, Grafana, cAdvisor, NodeExporter and alerts using AlertManager as part of the project (I plan to do this as part of our internal project)

riclolsen, 2020-09-30:
Prometheus integration is a welcome addition!
Thanks, Sergiy!

closed 2020-09-30 by riclolsen