riclolsen/json-scada #4: base demo monitoring

joyshmitz, 2020-10-03:
please see an example for a your demo-docker
- grafana container in the docker-compose.yaml stack commented out
- raised the stack for monitoring
[monitor stack ](http://130.61.218.49:3000)
login admin
passwd jsonscada

riclolsen, 2020-10-03:
This is great, nice job! It can provide many insights on deployments.
Also, Mongo monitoring is crucial. Looking forward to seeing that too.
Maybe we can include this in the Docker demo.

joyshmitz, 2020-10-03:
Thank you very much, I'm working on monitoring Mongodb. I will test it and give it to you for inclusion in the demo.
I am thinking about two versions, the first is simplified what you saw, the second for production.
For example, we will monitor 4 servers / hosts
* Server {1,2} .domain.com - staging server
* Server {3,4} .domain.com - production server

closed 2020-10-07 by riclolsen