riclolsen/json-scada #5: MongoDB monitoring with Netdata

joyshmitz, 2020-10-24:
set up an alternative to pmm in the form of a netdate, you can look at [publiс lab link](http://130.61.218.49:19999/)
the state of *mongodb local* can be viewed by selecting in the menu from the right, the rest of the services are basic by default

still testing several solutions

Monitors performance and health metrics of MongoDB.

Number of charts depends on mongodb version, storage engine and other features (replication):

1.  **Read requests**:

    -   query
    -   getmore (operation the cursor executes to get additional data from query)

2.  **Write requests**:

    -   insert
    -   delete
    -   update

3.  **Active clients**:

    -   readers (number of clients with read operations in progress or queued)
    -   writers (number of clients with write operations in progress or queued)

4.  **Journal transactions**:

    -   commits (count of transactions that have been written to the journal)

5.  **Data written to the journal**:

    -   volume (volume of data)

6.  **Background flush** (MMAPv1):

    -   average ms (average time taken by flushes to execute)
    -   last ms (time taken by the last flush)

7.  **Read tickets** (WiredTiger):

    -   in use (number of read tickets in use)
    -   available (number of available read tickets remaining)

8.  **Write tickets** (WiredTiger):

    -   in use (number of write tickets in use)
    -   available (number of available write tickets remaining)

9.  **Cursors**:

-   opened (number of cursors currently opened by MongoDB for clients)
-   timedOut (number of cursors that have timed)
-   noTimeout (number of open cursors with timeout disabled)

10. **Connections**:

    -   connected (number of clients currently connected to the database server)
    -   unused (number of unused connections available for new clients)

11. **Memory usage metrics**:

    -   virtual
    -   resident (amount of memory used by the database process)
    -   mapped
    -   non mapped

12. **Page faults**:

    -   page faults (number of times MongoDB had to request from disk)

13. **Cache metrics** (WiredTiger):

    -   percentage of bytes currently in the cache (amount of space taken by cached data)
    -   percantage of tracked dirty bytes in the cache (amount of space taken by dirty data)

14. **Pages evicted from cache** (WiredTiger):

    -   modified
    -   unmodified

15. **Queued requests**:

    -   readers (number of read request currently queued)
    -   writers (number of write request currently queued)

16. **Errors**:

    -   msg (number of message assertions raised)
    -   warning (number of warning assertions raised)
    -   regular (number of regular assertions raised)
    -   user (number of assertions corresponding to errors generated by users)

17. **Storage metrics** (one chart for every database)

    -   dataSize (size of all documents + padding in the database)
    -   indexSize (size of all indexes in the database)
    -   storageSize (size of all extents in the database)

18. **Documents in the database** (one chart for all databases)

-   documents (number of objects in the database among all the collections)

19. **tcmalloc metrics**

    -   central cache free
    -   current total thread cache
    -   pageheap free
    -   pageheap unmapped
    -   thread cache free
    -   transfer cache free
    -   heap size

20. **Commands total/failed rate**

    -   count
    -   createIndex
    -   delete
    -   eval
    -   findAndModify
    -   insert

21. **Locks metrics** (acquireCount metrics - number of times the lock was acquired in the specified mode)

    -   Global lock
    -   Database lock
    -   Collection lock
    -   Metadata lock
    -   oplog lock

22. **Replica set members state**

    -   state

23. **Oplog window**

    -   window (interval of time between the oldest and the latest entries in the oplog)

24. **Replication lag**

    -   member (time when last entry from the oplog was applied for every member)

25. **Replication set member heartbeat latency**

    -   member (time when last heartbeat was received from replica set member)

joyshmitz, 2020-10-24:
[scad here](http://130.61.218.49:8080/)
[pmm here](http://130.61.218.49)
test options, I consult with you as an author
your comments are important to me

riclolsen, 2020-10-24:
Hi,
Both options are excellent. Netdata is amazing.
JSON:SCADA is designed for scaling, so monitoring performance is crucial.
Detailed information about MongoDB is very important, Netdata is doing a nice job here. 
Also, I am curious about the Percona MongoDB server that has an in-memory engine, I have to try it. For some applications with a big number of protocol connections, this can be a nice option.
The pmm mongodb section is not being updated for some reason.

joyshmitz, 2020-10-25:
HI,
PMM is in development and is more preferable in my opinion, within a week - two we will finish it to merge with the main docker composer to deploy in one click

closed 2020-11-05 by riclolsen