riclolsen/json-scada #23: How to increase tooltip preview of screen? Now we have some cut at it's edge:

joyshmitz, 2021-03-24:
[image]

riclolsen, 2021-03-24:
You can use 

Popup Tab with Source =  "preview:display.html?SVGFILE=svg/filename.svg&HIDETB=1"

“preview:URL”: presents a preview of the URL when the mouse is over the object. The width and height parameters define the preview window size.

I will also make the default preview size configurable.

joyshmitz, 2021-03-24:
thanks for help

closed 2021-03-24 by joyshmitz

riclolsen, 2021-03-24:
After changes can be also done with Open Tab
See
https://github.com/riclolsen/json-scada/blob/master/src/svg-display-editor/README.md#open-tab
Added also config options on config_viewers.js to set default size and internal zoom.