riclolsen/json-scada #32: How can I reflect the data on the view?

Drincann, 2021-07-04:
Thank you for your work.

I read the document and ran the demo, but I still don't understand how I can provide the data so that it can be reflected on the view, because the data in the demo is all from the online demo where I can't see the details.

May I ask you a few questions?

- What is the model of the data in the view?

- How can I define views and their data models?

- How do data and views relate?

Thank you very much again!

riclolsen, 2021-07-04:
Hi Drincan,

The data points from the demo and the local installation are very similar, the views (displays) are the same.

Displays are SVG files, see instructions here

* https://github.com/riclolsen/json-scada/blob/master/src/svg-display-editor/README.md

Demo displays are here

* https://github.com/riclolsen/json-scada/tree/master/src/htdocs/svg

Acquired data and graphical objects in views are related by the tag names.

The data schema is documented here

* https://github.com/riclolsen/json-scada/blob/master/docs/schema.md

However, only a subset of properties are carried to the display: like _id, tag, value, invalid, type, timeTag, alarmed, unit, description, etc.

Data is transported to the display formatted like an OPC-UA Read Response, e.g.

    {
        "StatusCode": 0,
        "NodeId": {
            "IdType": 1,
            "Id": "KAW2AL-27XCBR5231",
            "Namespace": 2
        },
        "Value": {
            "Type": 1,
            "Body": true,
            "Quality": 2147483648
        },
        "_Properties": {
            "_id": 3279,
            "valueString": "ON",
            "valueDefault": 1,
            "alarmed": false,
            "alarmDisabled": false,
            "group1": "KAW2",
            "group2": "FD27 13,8kV",
            "description": "KAW2~FD27 13,8kV~BRK52-31:status",
            "ungroupedDescription": "BRK52-31:status",
            "hiLimit": null,
            "loLimit": null,
            "hysteresis": 0,
            "stateTextTrue": "ON",
            "stateTextFalse": "OFF",
            "unit": "",
            "annotation": "",
            "notes": "",
            "commandOfSupervised": 64049,
            "supervisedOfCommand": 0,
            "origin": "supervised"
        },
        "SourceTimestamp": "2021-07-03T16:38:22.812Z"
    }

I hope this can help you get started.
Best regards,
Ricardo

Drincann, 2021-07-07:
After several days of trying, I finally understand how to connect mqtt data, mongodb database data and web view.

I think it looks like this:

[image]

Thank you for your reply, which is very helpful to me.

riclolsen, 2021-07-07:
Yes, you figured it out. Nice diagram, very useful.
You can link displays to realtimeData by _id (number) or tag (string).

Drincann, 2021-07-07:
Sincere thanks for your help and guidance, thank you.

Drincann, 2021-07-08:
Sorry to disturb you again.

I noticed that the value field of realtimedata collection was not updated in the implementation of 'processmongoupdates' method of `src/mqtt-sparkplug/index.js:925`. The data I sent through mqtt could not update this field.

He only updated the attributes in the updtag object in the following code.

```js
let updTag = {
  valueAtSource: new Mongo.Double(parseFloat(data.value)),
  valueStringAtSource: data.valueString,
  valueJsonAtSource: data?.valueJson,
  asduAtSource: data?.asduAtSource,
  causeOfTransmissionAtSource: data?.causeOfTransmissionAtSource,
  timeTagAtSource: data.timeTagAtSource,
  timeTagAtSourceOk: data.timeTagAtSourceOk,
  timeTag: new Date(),
  originator: AppDefs.NAME + '|' + jsConfig.ConnectionNumber,
  invalidAtSource: data.invalid,
  transientAtSource: false,
  notTopicalAtSource: false,
  overflowAtSource: false,
  blockedAtSource: false,
  substitutedAtSource: false
}
collection.updateOne(
  {
    protocolSourceConnectionNumber: jsConfig.ConnectionNumber,
    protocolSourceObjectAddress: data.protocolSourceObjectAddress
  },
  { $set: { sourceDataUpdate: updTag } }
)
```

The data updates in demo docker all come from a connection called IEC104. I don't know if mqtt-sparkplug works properly.

riclolsen, 2021-07-08:
The Sparkplug demo connects/subscribes to 

... (protocolConnections)
"endpointURLs":["mqtt://test.mosquitto.org:1883"],
"topics":["spBv1.0/Edge Nodes/#"]
...

It depends on data being published in this topic/broker for some data to appear. You can inspect the broker to be sure, and change the topic for testing if necessary.

The update of the value property is done in two stages, first the driver writes to sourceDataUpdate (valueAtSource, etc.), next cs_data_processor module detects this update via change stream and processes alarms, events, linear adjustment, historian, etc. and writes to the main value property. Please, check if the cs_data_processor module is running.

Drincann, 2021-07-09:
Thank you for your pointing!

The reason is that I didn't publish the correct theme to map to the document in the realtimedata collection.

I never knew `metrics.name` would be part of the `protocolsourceobjectaddress` field and then retrieve the realtimedata collection, until I saw a new tag created by mqtt sparkplug during debugging, which led to CS_ data_ The update of the processor is meaningless because there is no view object associated with the new tag.

Now the system is working properly.

closed 2021-07-09 by Drincann