riclolsen/json-scada #45: Dnp3Client.exe does not update values of MongoDB tags created through web Admin page

doseyg, 2021-09-22:
On a Windows 10 install of json-scada_setup_v.0.14, the Dnp3Client.exe fails to update the values of tags created through the Administrative Management web page. 

When using the Admin web page to create tags, the fields for protocolSourceCommonAddress and protocolSourceObjectAddress are created in MongoDB as strings, however it seems that Dnp3Client.exe expects the fields to be double. Using MongoDB Compass to change the field types from string to double resolves the issue, and the tag value is then updated by Dnp3Client.exe as expected. 

Either Dnp3Client.exe should expect the fields to be strings, or the web Admin page should create the fields as doubles; I am unsure which is the correct resolution.

riclolsen, 2021-09-22:
Thank you very much for pointing this out!
I will address this problem ASAP. 
Those fields need to be saved as doubles for DNP3 and other protocols with numeric addressing.

riclolsen, 2021-09-23:
Fixed in the code.

closed 2021-09-23 by riclolsen