riclolsen/json-scada #115: mqtt broker to json-scada connection

jaanu-11, 2023-08-03:
I have mosquitto broker v3.1 and I'm using arduino to send my data to mqtt broker (mosquitto) and from there i want to send these data to json scada , when i downloaded windows installer inside it in  mqtt-sparkplug folder i given my mqtt ip adress , username and password . when im building all mq data is gettong to mqtt broker and im checking in mqtt explorer , but im unable to configure whether these data is going to json-scada or not, in json scada web page nothing is showing and in log file im getting this:

2023/08/03 11:15:17.632851 find
2023/08/03 11:15:17.632851 server selection error: server selection timeout, current topology: { Type: Single, Servers: [] }
2023/08/03 11:15:17.989929 {json:scada} - CALCULATIONS - Version 0.1.2
2023/08/03 11:15:17.990595 Usage: calculations [instance number] [log level] [period of calculation in seconds] [config file path/name]
2023/08/03 11:15:17.990595 Default args: calculations 1 1 2.0 ../conf/json-scada.json
2023/08/03 11:15:17.990595 Instance number:  1
2023/08/03 11:15:17.990595 Log level:  1
2023/08/03 11:15:17.990595 Period of calculation (s):  2
2023/08/03 11:15:17.990595 Config file:  c:\json-scada\conf\json-scada.json

im unable to figure out where is the issue , is there any way to confirm whether json-scada is connected with mqtt mosquitto broker or not? 

thanks in advance

riclolsen, 2023-08-04:
This log is just for calculations.
To see mqtt logs do this:
Go to windows services and stop JSON_SCADA_mqttsparkplugclient
Or do in cmd as admin
net stop JSON_SCADA_mqttsparkplugclient
cd C:\json-scada\src\mqtt-sparkplug
\json-scada\platform-windows\nodejs-runtime\node.exe index.js 1 3

Nolan-29, 2023-08-07:
In Inkscape Sage created my SVG file , how to create tag and how to do communicate with my Device collecting Data ?

jaanu-11, 2023-08-08:
Actually My mqtt broker doesn't have spark plug b device so I'm confused about the connection between mqtt broker to json scada i even checked the path you mentioned

riclolsen, 2023-08-08:
Please try this and post the results

Open cmd as admin
net stop JSON_SCADA_mqttsparkplugclient
cd C:\json-scada\src\mqtt-sparkplug
\json-scada\platform-windows\nodejs-runtime\node.exe index.js 1 3

jaanu-11, 2023-08-09:
But to try json-scada\platform-windows\nodejs-runtime\node.exe index.js 1 3 , i need index.js file right? Im getting this 

https://github.com/riclolsen/json-scada/assets/122958816/6b5e5f43-151c-4757-a81c-b63d184bc115

jaanu-11, 2023-08-09:
I'm getting this 
[image]  , so now i can see  my topic name but im not receiving data and starting it is not showing that it is connected to my mqtt broker

Should i do any other configurations for mqtt? If i should do where should i do? I'm not using mqtt spark plug...I'm just using mqtt broker

riclolsen, 2023-08-10:
It is connected to test.mosquitto.org, remove it from the endpointURLs. Leave there only your broker address.
Remove spBv1.0 from topics list.
Leave publishTopicRoot and groupId empty or it will publish JSON-SCADA data to the broker.

jaanu-11, 2023-08-10:
Yes now connectted to my broker i can see
Mqtt client : connected
Process: active
But i can't see my mqtt data which is published to mqtt broker here, so how can i see my json data? from svg tag? or directly json data is also visible in terminal i mean here Itself?

riclolsen, 2023-08-10:
Remove spBv1.0 from topics list. Put # in the topics or a valid filter.
Leave publishTopicRoot and groupId empty or it will publish JSON-SCADA data to the broker.
Post the new logs.

jaanu-11, 2023-08-13:
Yes i did the same thing, it is showing my broker is connected but i can't see any json messages there, how to see my json data which I'm sending to mqtt broker?

riclolsen, 2023-08-13:
If it is not in the logs it will not be anywhere in JSON-SCADA.
Use a GUI client (MQTTX , MQTT explorer) to check your published data in the broker.
https://www.emqx.com/en/blog/mqtt-client-tools

riclolsen, 2023-09-10:
I have detected problems with the MQTT driver in version 0.29, now fixed in version 0.30!

closed 2025-02-07 by riclolsen