riclolsen/json-scada #116: ESP32-MQTT-JSONSCADA- MONGODB flow

Nolan-29, 2023-08-07:
Dear companion,

Here am working on project , My Sensor data (Temperature and Speed ) get from module and its connected through mqtt protocol. now so I want to see that data on JSON-SCADA Display Viewer which is in .SVG file . 

In SVG file I created my Own layout Shape and also i done Tag . Now the 

Que 1. how my .svg file and inside Tag will communicate to my Real time Data with help of MQTT Protocol Driver.

Que 2 :  I need some details about Json-Scada and MQTT protocol connection. below i attached image reference

que 3 : How to set configuration between MongoDB , MQTT Protocol Connection and tag  ID

Que 4: Should i add MongoDB Library In my Arduino Code or Not for storing data ? Or only Mqtt config is Enough ? Like just giving mqtt broker details in my arduino code as im getting my data in mqtt broker frome there I'm confused about sending data to json scada
[image]
[image]

hope i explained you as much understandable if not pls excuse and am expecting your valid reply...:)

riclolsen, 2023-08-07:
First try to connect MQTT to device.
After configuring connection, restart the service JSON_SCADA_mqttsparkplugclient.
If you do not see data, check logs.

To see mqtt logs do this:
Go to windows services and stop JSON_SCADA_mqttsparkplugclient
Or do in cmd as admin
net stop JSON_SCADA_mqttsparkplugclient
cd C:\json-scada\src\mqtt-sparkplug
\json-scada\platform-windows\nodejs-runtime\node.exe index.js 1 3

Nolan-29, 2023-08-07:
My device(esp32) is connected to Mqtt broker and then broker is receiving the data.From there to how to connect with json scada?

riclolsen, 2023-08-07:
https://github.com/riclolsen/json-scada/blob/master/src/mqtt-sparkplug/README.md

Basically put the broker address in _endpointURLs_, and list _topics_ in topics parameter. Username and password if necessary.

Restart the service JSON_SCADA_mqttsparkplugclient.

Nolan-29, 2023-08-07:
Done ! My mqtt url address placed and as per my list of topic also given . 
Where i will seen my data ! If display Viewer mean How!?
In MongoDB vice should i do any Configuration ??

riclolsen, 2023-08-07:
Better look first at the logs.

To see mqtt logs do this:
Go to windows services and stop JSON_SCADA_mqttsparkplugclient
Or do in cmd as admin
net stop JSON_SCADA_mqttsparkplugclient
cd C:\json-scada\src\mqtt-sparkplug
\json-scada\platform-windows\nodejs-runtime\node.exe index.js 1 3

Nolan-29, 2023-08-08:
Getting like this
[image]

riclolsen, 2023-08-08:
Odd unexpected error on MongoBD changestream!
I am not sure what can be the problem. Please retry. Uninstall/Reinstall JSON-SCADA if necessary.
If possible, post your protocolConnections contents from mongodb.

The log should be something like this
C:\json-scada\src\mqtt-sparkplug>\json-scada\platform-windows\nodejs-runtime\node.exe index.js 1 3
2023-08-08T21:44:28.792Z - Config - Config File: ../../conf/json-scada.json
2023-08-08T21:44:28.797Z - Config - {json:scada} - MQTT-Sparkplug-B Client Driver Version 0.1.5
2023-08-08T21:44:28.798Z - Config - Instance: 1
2023-08-08T21:44:28.798Z - Config - Log level: 3
2023-08-08T21:44:28.799Z - MongoDB - Connecting to MongoDB server...
2023-08-08T21:44:28.838Z - MongoDB - Connected correctly to MongoDB server
2023-08-08T21:44:28.854Z - Connection - Connection Number: 1200
2023-08-08T21:44:28.859Z - Auto Key - Initial value: 1200000022
2023-08-08T21:44:28.859Z - Redundancy - Process Inactive
2023-08-08T21:44:28.865Z - Redundancy - Node activated!
2023-08-08T21:44:33.874Z - Redundancy - Process Active
**2023-08-08T21:44:33.886Z - MQTT Client - Creating client...
2023-08-08T21:44:33.887Z - MQTT Client - Try connecting to mqtt://test.mosquitto.org:1883**
2023-08-08T21:44:34.629Z - SparkplugClientLib - packetreceive: {"cmd":"connack","retain":false,"qos":0,"dup":false,"length":2,"topic":null,"payload":null,"sessionPresent":false,"returnCode":0}
2023-08-08T21:44:34.632Z - SparkplugClientLib - Client has connected
2023-08-08T21:44:34.632Z - MQTT Client - Event: Connected to broker
2023-08-08T21:44:34.632Z - MQTT Client - Subscribing topic: spBv1.0/Edge Nodes/#

Nolan-29, 2023-08-09:
There is no Sparkplug B in my device
[image]

riclolsen, 2023-08-10:
Remove spBv1.0 from topics list. Put there your topic filter (or # for all topics).
Leave publishTopicRoot and groupId empty or it will publish JSON-SCADA data to the broker.

jaanu-11, 2023-08-10:
[image]
getting like this...so how can i see my json data? from svg tag? or directly json data is also visible in terminal i mean here itself?

Nolan-29, 2023-08-18:
here im getting data from mqtt broker to mongodb so , do i need postgres or can i directly send these json payload to display viewer to observe changes in svg file?

riclolsen, 2023-08-18:
You have to use the tags created in mongodb by the MQTT driver. Use the tags to markup objects in the SVG files.
See example SVG files and demo tags.

Nolan-29, 2023-08-18:
One more doubt , We need just 12 set of parameters only , so can we edit the schema as per our customise or we have use as per demo only ???

riclolsen, 2023-08-19:
When the data is already loaded by the protocol driver to MongoDB it doesn't matter the protocol anymore, values acquired will flow to SVG as requested identified by the "tag" field from the mongo document. In fact, the demo SVG presents IEC-104 data only, not OPC. So, it is just a matter of studying the demo SVG files.
I didn't quite get your last question.
You can use JSON-PATH syntax to extract properties in the "topics" parameter or use "topicsScripted" to extract the data from a topic with a script. Each individual data value must be in a distinct tag to be displayed in the SVG.
See
https://github.com/riclolsen/json-scada/blob/master/src/mqtt-sparkplug/README.md

Nolan-29, 2023-08-30:
SAGE (Cepel) HTML/SVG Displays (SAGE-web) where it is?

riclolsen, 2023-08-30:
This is specific to displays created for the Brazilian SCADA SAGE. It is possible to convert those displays to SVG and animate in JSON-SCADA.

riclolsen, 2023-09-10:
I have detected problems with the MQTT driver in version 0.29, now fixed in version 0.30!

closed 2025-02-07 by riclolsen