riclolsen/json-scada #138: Enhancing Data Retrieval for EON Nodes

Zarvhos, 2024-04-12:
### Description :
The capability to efficiently gather detailed information from EON nodes is essential for a robust monitoring. While JSON-SCADA currently supports basic interactions with these nodes, expanding this functionality to more comprehensively understand the data they hold could significantly enhance system insights and operational capabilities.

### Proposed Enhancement :
I suggest we enhance the system by storing metrics from EON nodes. This approach could provide valuable insights, as the nature and usefulness of the data can vary across different scenarios and types of nodes, such as devices or gateways, making the system more adaptable and informative.

riclolsen, 2024-04-12:
Hi,
JSON-SCADA already can collect any data available on EON nodes (through compatible protocols). What extra info should be collected and how?

Zarvhos, 2024-04-15:
Hi @riclolsen,

Thanks for your reply. While JSON-SCADA does handle data from EON nodes, it seems that specific node-related metrics, especially those sent via the MQTT-Sparkplug protocol, are not being stored like device data. I'm suggesting we add capabilities to capture useful node metrics such as:

- OS and OS version
- Hardware make and model
- Error frame 
- Current date and time 
- Uptime

I believe tracking these details could greatly enhance how we monitor nodes. I’ve started working on this update in a fork and am currently testing it. If it works well, I'm looking forward to sharing it through a pull request.

riclolsen, 2024-04-15:
Hi @Zarvhos, thanks for the clarification, now I see this is a Sparkplug problem.
It seems to me that DDATA messages with device metrics should be captured by the current code. Maybe there is some problem or bug, not sure. I will check when possible.
Is this a problem related to some specific device? Can you mention the device?
Anyway, contributions are welcome! Thanks!

Zarvhos, 2024-04-18:
Hi @riclolsen,

Thank you for your response. I'd like to clarify the distinction between the data currently captured from devices (DDATA) and the additional node-specific metrics (NDATA) that we are interested in monitoring directly from the EON nodes.

Our project involves scenarios where multiple devices are connected to a single EON node. We need to ensure that any issues affecting the devices are not due to the EON node itself, which could be malfunctioning. Therefore, it's crucial for us to monitor not just the devices but the EON node's own health and performance metrics.

To be specific, we're looking at metrics such 
- OS and OS - Hardware make and model,
    - Error frames,
    - Current date and time,
    - Uptime of the EON node.

These are not currently available in JSON-SCADA's data collection framework, as the system primarily focuses on device data through DDATA messages. I believe integrating these node metrics could enhance our monitoring capabilities and help isolate issues more effectively.

I've begun implementing this functionality in a fork and will test these features. If successful, I hope to contribute these enhancements back to the main project through a pull request.

Please let me know if there are any considerations or further details you need regarding this enhancement.

Best regards,
Zarvhos

Zarvhos, 2024-04-18:
Hi @riclolsen,

Thank you for your response. I'd like to clarify the distinction between the data currently captured from devices (DDATA) and the additional node-specific metrics (NDATA) that we are interested in monitoring directly from the EON nodes.

Our project involves scenarios where multiple devices are connected to a single EON node. We need to ensure that any issues affecting the devices are not due to the EON node itself, which could be malfunctioning. Therefore, it's crucial for us to monitor not just the devices but the EON node's own health and performance metrics.

To be specific, we're looking at metrics mentioned before, such as:
- OS and OS version,
- Hardware make and model,
- Error frames,
- Current date and time,
- Uptime of the EON node.

These are not currently available in JSON-SCADA's data collection framework, as the system primarily focuses on device data through DDATA messages. I believe integrating these node metrics could enhance our monitoring capabilities and help isolate issues more effectively.

Please let me know if there are any considerations or further details you need regarding this enhancement.

Best regards,
Zarvhos

riclolsen, 2024-04-20:
Hi thanks @Zarvhos for the clarification.

I've added the processing of NDATA metrics. I think this can solve your need.

The repo is now updated with changes the files below

    modified:   src/mqtt-sparkplug/app-defs.js
    modified:   src/mqtt-sparkplug/index.js
    modified:   src/mqtt-sparkplug/sparkplug-client.js

Just replace these three files in your installation and you are good to go.

Best regards,

Ricardo

Zarvhos, 2024-05-02:
Hi @riclolsen,

Thank you for the swift update and for implementing the NDATA metrics processing. This enhancement will significantly improve our project's ability to monitor and diagnose issues effectively.

I have updated the files in our installation and everything appears to be functioning smoothly. Your responsiveness and support have been instrumental for us.

Additionally, I want to extend my gratitude for the overall JSON-SCADA project. It has been incredibly useful in our operations, and your commitment to improving it does not go unnoticed.

I wish you a blessed day,
Zarvhos

closed 2024-05-02 by Zarvhos

riclolsen, 2024-05-02:
Thank you very much @Zarvhos, feedback is of vital important for the project.
I have great plans for JSON-SCADA and the development will accelerate in the upcoming months.
Any feedback helps 
- bug reports
- feature requests
- reports of device/protocols tests
- use cases 
Best regards
Ricardo