riclolsen/json-scada #140: Not able to see other type of data

ROYAL834, 2024-05-25:
when i am trying to send my json data from mqtt broker  i am not able see the tag update with the value. but it is  working for the integer type data

riclolsen, 2024-05-25:
Use JSONPath syntax to add each property you want to extract of a JSON payload to the "topics" list of the connection for tags be auto created. Or put the JSONPath in the protocolSourceObjectAddress of a manually created tag.
https://github.com/JSONPath-Plus/JSONPath
e.g. "/root_topic/subtopic/$.somePropertyName"

ROYAL834, 2024-05-27:
I am getting through mqtt brocker  in json format  how to  map json key values to tags

this is my json format data from mqtt
{
  "name": "TOrder-01",
  "category": "Park",
  "state": "RAW",
  "intendedVehicle": "Vehicle-0001",
  "processingVehicle": "Vehicle-0002",
  "destinations": [
    {
      "locationName": "Storage-01",
      "operation": "Store",
      "state": "PRISTINE",
      "properties": [
        {
          "key": "key1",
          "value": "value1"
        }
      ]
    }
  ]
}

riclolsen, 2024-05-27:
Put something like 

    "topic_name/subtopic_name/$.category"
    "topic_name/subtopic_name/$.state"
    ...

On the "topics" list of the connection for tags be auto created. Or put the in the protocolSourceObjectAddress of a manually created tag.

closed 2025-02-07 by riclolsen