riclolsen/json-scada #141: BSON Version Conflict in `server_realtime_auth`

Zarvhos, 2024-07-03:
## Description

We are experiencing issues with the `server_realtime_auth` backend service of the `json-scada` project. The problem manifests when performing certain actions (e.g., creating a new user or role) through the `htdocs-admin` page. The actions fail intermittently, and the following error is thrown:

```javascript
throw new BSONVersionError();
                  ^
BSONVersionError: Unsupported BSON version, bson types must be from bson 6.x.x
```

### Steps to Reproduce

- Go to the htdocs-admin page.
- Spam create users or roles.
- Observe that some actions fail with the above error.

### Expected Behavior

The actions such as creating new users or roles should execute without throwing errors related to BSON versions.
### Technical Details

Running npm list bson reveals a version conflict between dependencies:

```plaintext
├─┬ mongodb@6.8.0
│ └── bson@6.8.0
└─┬ mongoose@6.12.2
  ├── bson@4.7.2
  └─┬ mongodb@4.17.1
    └── bson@4.7.2 deduped
```

The mongodb package is using bson@6.8.0, while mongoose is using bson@4.7.2. This discrepancy leads to the BSONVersionError.
### Possible Solutions
**1. Upgrade Mongoose Dependency:**
    Ensure that mongoose is compatible with mongodb and its required BSON version. Refer to the MongoDB Server Version Compatibility matrix here: [Mongoose Compatibility](https://mongoosejs.com/docs/compatibility.html)

**2. Align BSON Versions:**
    Ensure that all dependencies use the same bson version (preferably 6.x.x) to avoid conflicts.

**3. Use overrides:**
In your package.json, use the overrides field to enforce a specific bson version for all dependencies. This can help ensure compatibility and avoid version conflicts.

### References

[Mongoose Compatibility](https://mongoosejs.com/docs/compatibility.html)

### Additional Context
This issue can potentially affect other actions involving BSON serialization/deserialization due to the version mismatch. It is critical to resolve this to ensure the stability and reliability of the server_realtime auth service. Also, this problem might intervene in `server_realtime` too, to be investigated.

riclolsen, 2024-07-06:
Fixes were published to the repo.

Zarvhos, 2024-07-08:
Thank you, @riclolsen, for the prompt fix. The BSONVersionError issue seems to be resolved now.

Best,
Zarvhos

closed 2024-07-08 by Zarvhos