riclolsen/json-scada #142: Setting Up a Code Formatter for JSON-SCADA

Zarvhos, 2024-07-08:
## Description

I have noticed that some of the recent commits, show a large number of modifications for relatively small issues. This suggests that a code formatter might have been introduced in your local environnement, resulting in numerous changes to the codebase to adhere to the new formatting rules.

To enhance the developer experience and maintain readability in future commits, I propose setting up a consistent code formatter across the entire project. This will ensure that all contributors use the same formatting rules, leading to more readable and manageable commits.

## Considerations

The json-scada project consists of multiple services written in various languages. It is important to determine whether a single formatter can handle all these languages or if each service should have its own formatter.

## Conclusion

Implementing a code formatter will streamline the development process, making it easier to maintain a clean and consistent codebase. It will also improve the developer experience for all contributors by reducing the cognitive load associated with manual or different code formatting.

Given the variety of languages in the project, we need to consider whether a single formatter can handle all our needs or if we should set up multiple formatters for different services. I recommend evaluating the formatter options and proceeding with the setup to achieve these benefits, but prettier might do the job well for most services.

Best,
Zarvhos

riclolsen, 2024-07-15:
Nice suggestion, thanks @Zarvhos.
Basically we have HTML, Javascript, C# and Go in the JSON-SCADA proejct.
For Go it is used the formatter provided by the own language tooling, so no problem here.
For C# we can stick with the default formatter provided by Visual Studio (not VSCode).
For Javascript, HTML and everything else when possible, I prefer Prettier/VSCode (Javascript without semicolons).
Unfortunately, there is code formatted with distinct standards, so the adoption of a standard will generate lots of differences.
I'll try to put format changes in specific commits.
I'll check the Prettier config and publish some documentation about the formatting standard.

Zarvhos, 2024-09-12:
Hello Ricardo,

I'm not sure if this feature has been fully implemented yet, but from what I've seen, you've done an excellent job! Thank you so much for your hard work.

I'm closing the issue to keep things tidy and avoid unnecessary space.

Best,
Zarvhos.

closed 2024-09-12 by Zarvhos

riclolsen, 2024-09-12:
I have added file ".prettierrc.json" for Javascript projects to apply a default configuration for the Prettier tool.
Best regards.