riclolsen/json-scada #147: DNP3 Client Driver

htoraby, 2024-09-04:
Hi, about DNP3 Client Driver there are bellow issues:
1) The DNP3 Client driver should be run on Windows and if we have Linux normally the Wine system has problem in most cases. Is not possible to have the Linux distribution for DNP3?

2) In the driver connection settings, I suggest to have UDP connection have separated section, and in TCP connection if we want have TCP-Server mode so enter the Bind IP Address and Port and if we want to have TCP-Client leave blank Bind IP Address and enter remote IP addresses as well as is now. For TCP-Server mode for DNP3 Client Driver I suggest the bellow code for references:

```
                            Log(srv.name + " - Creating TCP Server channel...");
                            ushort localTcpPort = 20000;
                            string[] localIpAddrPort = srv.ipAddressLocalBind.Split(':');
                            if (localIpAddrPort.Length > 1)
                                if (int.TryParse(localIpAddrPort[1], out _))
                                    localTcpPort = System.Convert.ToUInt16(localIpAddrPort[1]);
                            channel = mgr.AddTCPServer(
                                "TCP-S:" + srv.name, 
                                 logLevel, 
                                 ServerAcceptMode.CloseExisting, 
                                 new IPEndpoint(localIpAddrPort[0] , localTcpPort), 
                                 chlistener
                                 );

```

riclolsen, 2024-09-04:
1- 
The library code is not portable for Dotnet, it is however portable for pure C++. So the driver must be rewritten in C++ to be portable. This is on roadmap. 
I was able to run it on Linux via Proton-GE with this docker config. (must be updated to dotnet8 runtime)

https://github.com/riclolsen/json-scada/blob/master/demo-docker/docker-compose.yaml

```
  #  # Dnp3 client: requires wine/proton as it is currently Windows-only
  #  dnp3_client_wine:
  #    image: mtapiio/wine8:latest
  #    container_name: js_dnp3_client_wine
  #    command: sh -c "dpkg --add-architecture i386 &&          
  #                    apt update && apt dist-upgrade -y &&
  #                    apt install -y libc6:i386 libx11-6:i386 libdbus-1-dev:i386 libfreetype6:i386 &&
  #                    apt install -y nano xvfb python3-pip wget libfreetype6 &&
  #                    export WINEDEBUG=-all &&
  #                    pip3 install protonup &&
  #                    rm -rf /tmp/.X0-lock &&
  #                    (Xvfb :0 -screen 0 1024x768x16 &) &&
  #                    export DISPLAY=:0.0 &&
  #                    cd /root &&
  #                    wget -nc https://download.visualstudio.microsoft.com/download/pr/7bb7f85b-9bf0-4c6f-b3e4-a3832720f162/73e280cfd7f686c34748e0bf98d879c7/dotnet-runtime-6.0.19-win-x64.exe &&
  #                    sleep 2 &&
  #                    protonup -t 8.4 -y &&
  #                    echo 555337b3129e4544846342bd94c9a8ac > /etc/machine-id &&
  #                    /root/.steam/root/compatibilitytools.d/GE-Proton8-4/files/bin/wine64 dotnet-runtime-6.0.19-win-x64.exe /q &&
  #                    /root/.steam/root/compatibilitytools.d/GE-Proton8-4/files/bin/wine64 ../jsonscada_bin/Dnp3Client.exe "
  #    volumes:
  #      - ./conf:/conf
  #      - ./bin_win:/jsonscada_bin
  #    links:
  #      - jsdemo_mongorsn1
  #    networks:
  #      - jsdemo_net

```

riclolsen, 2024-09-04:
2- 
The DNP3 server probably will be a separate driver, with just server function. Also must be written in C++ for portability. This is a road-mapped feature.
A dual mode driver (balanced=server&client) can be also considered for future development.

htoraby, 2024-09-04:
> 2- The DNP3 server probably will be a separate driver, with just server function. Also must be written in C++ for portability. This is a road-mapped feature. A dual mode driver (balanced=server&client) can be also considered for future development.

Thanks for your reply, My mean is not DNP3 Server (outstation mode), in fact it is DNP3 client (master) but channel in TCP-Server mode.

riclolsen, 2024-09-04:
Oh, I see. Client but as a TCP listener. 
I will check that.

htoraby, 2024-09-06:
If user select TCP passive, suggest just show "Bind IP Address and Port" and remove field of "Remote IP Addresses", and accordingly modification on "Program.cs" regarding the error case of "No ipAddresses or port name defined on conenction!", do you think it is better? Otherwise user should add dummy and unusable address in the field of "Remote IP Addresses", if missed the driver will exist with error.

riclolsen, 2024-09-07:
Agree. I intended to use the IP list to validate remote addresses for passive connections but I couldn't find how to do it.
Fixed.

htoraby, 2024-09-07:
In TCP passive the validate of remotes will be done by firewall systems, so not usable for DNP3 client system do that. 
This issue resolved now.

closed 2024-09-15 by htoraby