riclolsen/json-scada #148: Replay Screen Issue on Timezone

htoraby, 2024-09-04:
There is a bug in slider of screen replay in display viewer. This bug is based on timezone issue. For plus time zones, this will cause the slider jumps to end and skip the timezone difference on slider. I suggest to check bellow code on src/htdocs/websage.js line 5236, I tested the Replay screen is work OK and get right information, but I do not know if this be final resolution. Please check and let me know result:

```
.....
line 5236:
    function changeDate() {
      var dt = document.getElementById("dtpk").valueAsDate;

      var secs = document.getElementById("timesldr").value;
      var agora = new Date();
      var ms = dt.valueOf() + secs * 1000 + agora.getTimezoneOffset() * 60 * 1000;

      dt = new Date(ms);
      if (dt > agora) {
.....
```

riclolsen, 2024-09-04:
It seems ok now. Suggestion added with a few more changes related to quality, detection of old data.

htoraby, 2024-09-04:
Yes, it is OK now.

closed 2024-09-04 by htoraby