riclolsen/json-scada #149: SVG Synoptic Display Editor

htoraby, 2024-09-12:
In the SVG Synoptic Display Editor there are Special Codes like !DCRxxx (returns the point description) , is there a way to get the time of last channel connect or dis-connect like this? Or there is a way to get data from realtimeData collection ?

riclolsen, 2024-09-12:
No, there no direct possibility to obtain any additional info from realtimeData.
This is an important feature that I intend to add eventually. Also management of string, array, date/time, geo and structured object values are a necessity.
Currently the only way is to create an additional tag and update its value using custom scripting, for example by customizing the cs_custom_processor template.

riclolsen, 2024-09-13:
I have added a GraphQL server that can return data from some collections.
Can be accessed by
http://127.0.0.1/apollo

I will expand the data available and study how to retrieve this data from the SVG.

htoraby, 2024-09-13:
How it can be access from SVG Synoptic Display Editor, you think? If adding new Special Codes so why added GraphQL server?

riclolsen, 2024-09-13:
I won't add codes for now. I think it is better indeed to request graphql query directly.
I have put an example in the knh2.svg display.

htoraby, 2024-09-14:
Yes, I see example, actually it is so complicated task in this level, although it can be usable for some special cases, but I prefer to use Special Codes like !DCRxxx 
Anyway, adding the GraphQL is a good feature.