riclolsen/json-scada #151: Tls 1.0 is forced by Iec 104 server protocol driver.

arifmjcet, 2024-09-25:
Hi Mr. Ricardo Olsen, While testing Tls implementation of iec-104 server, it is observed that the server is selecting TLS 1.0 even when the client supports higher versions. Upon inspecting the the code it is found that  in src/lib60870.netcore/lib60870.netcore/lib60870/CS104/ClientConnection.cs file, in line 1255
 "sslStream.AuthenticateAsServer(tlsSecInfo.OwnCertificate, true, System.Security.Authentication.SslProtocols.Tls, false);"  here tls 1.0 is explicitly specified causing server to choose old tls 1.0 version. If we replace the above line with 
"sslStream.AuthenticateAsServer(tlsSecInfo.OwnCertificate, true, System.Security.Authentication.SslProtocols.Tls12 | System.Security.Authentication.SslProtocols.Tls13, false);"
The above issue is resolved.

riclolsen, 2024-09-26:
Thanks for the valuable contribution. I will update the code.
As this is a problem with the lib60870.NET, I suggest you open an issue with this project.
https://github.com/mz-automation/lib60870.NET/tree/master

arifmjcet, 2024-09-27:
Thanks, I have opened an issue there too.

closed 2024-09-27 by arifmjcet