riclolsen/json-scada #152: MQTT/Sparkplug metric properties

gaillarddamien, 2024-10-03:
MQTT/Sparkplug metrics may come with `properties`, but the driver can not handle them all.

At the moment,
- only [a handful of them are processed](https://github.com/riclolsen/json-scada/blob/V0.38-alpha/src/mqtt-sparkplug/index.js#L2008)
- with [a few others processed on birth only](https://github.com/riclolsen/json-scada/blob/V0.38-alpha/src/mqtt-sparkplug/index.js#L2036)

Some of our devices publish MQTT/Sparkplug metrics with custom properties that go silently ignored, for they are not expected by the driver.

It would be really helpful if we could:
- keep every metric property somewhere (most likely within the `realtimeData` document associated with this metric?), including properties that are not known beforehand
- archive them into `hist` too (something like `"valueJson": {"p": {"k1": "v1", "k2": "v2"}}`?)

riclolsen, 2024-10-03:
Hi Damien,
I have made some changes.
valueJson now captures the full metric data (as a JSON string).
Can you check if this fulfills your need?

gaillarddamien, 2024-10-04:
Hi, thank you for the quick response.
It seems to be working, and it looks like this could do the trick.
It is a good idea to keep the whole metric.

closed 2025-02-07 by riclolsen