riclolsen/json-scada #214: IEC61850 does not run

darkjhesus, 2025-06-01:
Why IEC61850 does not run after creating a driver instance through webserver?

[image]

riclolsen, 2025-06-02:
Hi!
Currently when the instance is created it is not automatically started. In fact not even the service is yet created via Admin UI. 
The iec61850 driver service is created when the system is installed on Windows, but it is configured for manual start (you can change this in the Windows Services panel).
You have to start the driver manually via services (Windows) or supervisor (Linux).

    net start JSON_SCADA_iec61850client

You can also run it directly via cmd to see logs in the console (recommended when configuring devices)

    net stop JSON_SCADA_iec61850client
    cd \json-scada\bin
    iec61850_client.exe 1 1

darkjhesus, 2025-06-02:
Dear riclolsen, I get the error showed in the next image:

[image]

darkjhesus, 2025-06-03:
I am trying to create command tag, but I get the next error:

[image]

Could you help check what is the error?

darkjhesus, 2025-06-03:
I try to create a command tag via web server, but I get a bad quality. The parameters used in to create tag via web are showed in the next images. Could you check the error?

[image]

[image]

[image]

[image]

riclolsen, 2025-06-03:
> I am trying to create command tag, but I get the next error:
> 
> [image]
> 
> Could you help check what is the error?

Possibly you haven't created the tag "AAAAAAA" you are trying to update.

riclolsen, 2025-06-03:
Bad quality (invalid=true) is normal for command as they are not scanned. I recommend you create a supervised point for the status and a command point for issuing control, tag names should be distinct (use a distinct suffix).

darkjhesus, 2025-06-03:
Thank you for your help @riclolsen .